Microsemi/switchtec-user #48: switchtec fw-update - unknown error

danglus, 2018-10-31:
Hi,
  Any idea what may cause the following failure? (I'm trying to burn from Linux the same image that I burn from serial using switchlink).

```[vastdata@localhost ~]$ sudo switchtec-user/switchtec fw-info switchtec0
Currently Running:
  IMG Version: 1.09 B063
Active Partition:
  BOOT 	Version: 1.09 B063	CRC: e8d37467   (RO)
  MAP 	Version: 1.09 B063	CRC: 523ced6e
  IMG  	Version: 1.09 B063	CRC: 23d057bf	(Running)
  CFG  	Version: 1.09 B063	CRC: d1049752	(Running)
   	Multi Config 0
   	Multi Config 1 - Active
Inactive Partition:
  IMG  	Version: 1.09 B063	CRC: 23d057bf
  CFG  	Version:         	CRC: ffffffff
[vastdata@localhost ~]$ sudo switchtec-user/switchtec fw-update switchtec0 Sep18/vast_2.0_x96.pmc
Writing the following firmware image to switchtec0.
File:     vast_2.0_x96.pmc
Type:     DAT
Version:  1.09 B063
Img Len:  0x9c18
CRC:      0xbef280fc
Do you want to continue? [y/N] y
 100% [================================================================================================================] ETA:  0:00:00   37kB/s
firmware update: Unknown Error (1)
[vastdata@localhost ~]$
```

danglus, 2018-10-31:
I used the -f option and it worked.
   go figure

closed 2018-11-03 by danglus

wesleywesley, 2019-03-21:
@danglus 
This is a firmware bug we root caused.

The reason is:
After the whole firmware image download completely, the FW DOWNLOAD STATUS does not enter into one of the expected state (0x7, 0x8, 0x9), but keep at unexpected (0x1).

•0x01 - Download in progress, no error
•0x07 - Success - Download is completed and new image is activated
• 0x08 - Success - New firmware image has been downloaded successful successfully, if Image
Activation Option = 1, the new image needs to be toggled to active.
• 0x09 - Success - New configuration image has been downloaded successful successfully, if Image
Activation Option = 1, the new image needs to be toggled to active.