Microsemi/switchtec-user #54: NTB Driver question

XIO-Engineer, 2018-12-05:
How does the switchtec driver translate the addresses between two CPU’s or two hosts?  How do the two switches figure out the address translations between the first switch’s CPU’s/host’s addresses on the upstream ports, through the NT Only port, to the second switch’s NT Only port, back up to the second switch’s upstream port?

wesleywesley, 2018-12-06:
This issue should be opened under repo: Switchtec-kernel.
Pls closed it and reopen it on Switchtec-kernel.

wesleywesley, 2019-03-21:
@XIO-Engineer 

What you described is crosslink case.
General speaking, the address translation is like following procedure:
1. Host0 access its ntb bar of partiton with USP in self switch 0
2. The access lead to address translation from host0 space to peer switch 1's nt only bar space
3. Then the second address translation happened from switch 1's nt only bar address to switch 1's USP exist partition's space

Regard,
Wesley