Microsemi/switchtec-user #73: suggested patch to allow display of ltssm minor state

vermilionink, 2019-02-13:
Sir,

In release 1.1 rc1, switchtec.c, in the function call to retrieve port status, (command: "switchtec status"), one can change the second argument of the call to ltssm_str to 1, to enable display of ltssm minor states.
s[p].ltssm_str = ltssm_str(s[i].ltssm, 0);

However, the correct ltssm minor state is not displayed. This is due to the ltssm variable declaration in switchtec.h not being long enough to hold both the major and the minor information. When the change below is made, switchtec status shows the correct minor state.

diff --git a/inc/switchtec/switchtec.h b/inc/switchtec/switchtec.h
index 39fc567..c9d8a06 100644
--- a/inc/switchtec/switchtec.h
+++ b/inc/switchtec/switchtec.h
@@ -109,7 +109,7 @@ struct switchtec_status {
 	unsigned char neg_lnk_width;	//!< Negotiated link width
 	unsigned char link_up;		//!< 1 if the link is up
 	unsigned char link_rate;	//!< Link rate/gen
-	unsigned char ltssm;		//!< Link state
+	uint16_t ltssm;			//!< Link state
 	const char *ltssm_str;		//!< Link state as a string 
 	char *pci_dev;			//!< PCI BDF of the port

Could you please confirm the change and roll it into the mainline code?

wesleywesley, 2019-02-13:
@vermilionink 
Pls create a pull request with target branch set to devel.
We will go through the pull request procedure then.

regard,
Wesley

vermilionink, 2019-02-27:
created pull request #75. Thank you.

wesleywesley, 2019-03-07:
The pull request 77 is tracking this issue. So close this issue.

closed 2019-03-07 by wesleywesley