Microsemi/switchtec-user #76: redundant "-r" option of gui command

wesleywesley, 2019-03-07:
Hi, Guys,

GUI command invoke "switchtec_bwcntr_set_many()" firstly in gui_init() of gui_main(), it clear/reset the bw cntr as a by-product besides setup bw cntr mode (raw/payload).
For this reason, the followed "switchtec_bwcntr_many()" w/ reset is unnecessary.

From the user point of view, "-r" option confuse customer, because w/ or w/o this option, the results is identical. ("-r" option and button press of "r" at runtime are two different things)

Suggesting remove "-r" option gui command from its argument list.

Regard,
Wesley

lsgunth, 2019-03-07:
Oh, that's odd. If that's the case there's a bunch of cleanup and confusing code to be done there.

wesleywesley, 2019-03-08:
If you guys agree remove the -r option, I will try to generate a patch for review.