Microsemi/switchtec-user #80: C++ compilation fails

Floessie, 2019-03-12:
Hi,

when including `switchtec.h` from a C++ project the compilation fails due to the use of a C++ keyword for a parameter name.

Here's a patch to solve that:

```patch
diff --git a/inc/switchtec/switchtec.h b/inc/switchtec/switchtec.h
index d26bb6b..261ec69 100644
--- a/inc/switchtec/switchtec.h
+++ b/inc/switchtec/switchtec.h
@@ -607,8 +607,8 @@ struct switchtec_bwcntr_res {
 	  ingress;			//!< Bandwidth into the port
 };
 
-void switchtec_bwcntr_sub(struct switchtec_bwcntr_res *new,
-			  struct switchtec_bwcntr_res *old);
+void switchtec_bwcntr_sub(struct switchtec_bwcntr_res *new_cntr,
+			  struct switchtec_bwcntr_res *old_cntr);
 int switchtec_bwcntr_set_many(struct switchtec_dev *dev, int nr_ports,
 			      int * phys_port_ids,
 			      enum switchtec_bw_type bw_type);
diff --git a/lib/pmon.c b/lib/pmon.c
index a26f9bf..33f7ffb 100644
--- a/lib/pmon.c
+++ b/lib/pmon.c
@@ -331,21 +331,21 @@ int switchtec_evcntr_wait(struct switchtec_dev *dev, int timeout_ms)
 
 /**
  * @brief Subtract all the values between two bwcntr result structures
- * @param[in,out] new
- * @param[in] old
+ * @param[in,out] new_cntr
+ * @param[in] old_cntr
  *
- * \p new will have it's original values minus all the values in \p old
+ * \p new_cntr will have it's original values minus all the values in \p old_cntr
  */
-void switchtec_bwcntr_sub(struct switchtec_bwcntr_res *new,
-			  struct switchtec_bwcntr_res *old)
+void switchtec_bwcntr_sub(struct switchtec_bwcntr_res *new_cntr,
+			  struct switchtec_bwcntr_res *old_cntr)
 {
-	new->time_us -= old->time_us;
-	new->egress.posted -= old->egress.posted;
-	new->egress.nonposted -= old->egress.nonposted;
-	new->egress.comp -= old->egress.comp;
-	new->ingress.posted -= old->ingress.posted;
-	new->ingress.nonposted -= old->ingress.nonposted;
-	new->ingress.comp -= old->ingress.comp;
+	new_cntr->time_us -= old_cntr->time_us;
+	new_cntr->egress.posted -= old_cntr->egress.posted;
+	new_cntr->egress.nonposted -= old_cntr->egress.nonposted;
+	new_cntr->egress.comp -= old_cntr->egress.comp;
+	new_cntr->ingress.posted -= old_cntr->ingress.posted;
+	new_cntr->ingress.nonposted -= old_cntr->ingress.nonposted;
+	new_cntr->ingress.comp -= old_cntr->ingress.comp;
 }
 
 /**
```

Best,
Flössie

wesleywesley, 2019-03-18:
@Floessie 
Thx.

Since Switchtec-user is C based program, the C++ operator "new" has none impact on it.
When it come to C++ based case, adaptors could change by themselves, like what you does.

Regard,
Wesley

Floessie, 2019-03-18:
@wesleywesley 

> When it come to C++ based case, adaptors could change by themselves, like what you does.

What? Excuse me, but you refuse to change a single parameter name and want every user of the library using C++ to patch the sources? It's 2019...

This library – opposed to any other C library I've worked with in the past 15 years – cannot be used as is in C++ applications. Period.

lsgunth, 2019-03-18:
Yes, we should support using the library in C++. The patch is pretty benign so Wesley you should accept it. @Floessie, it would be easier if you sent a proper PR.

Logan

Floessie, 2019-03-18:
Hi Logan,

Thanks. I'll prepare a PR and close the issue here as soon as it's ready.

Best,
Flössie

Floessie, 2019-03-18:
See #82. Closing.

closed 2019-03-18 by Floessie