Microsemi/switchtec-user #84: CONTRIBUTING.md should describe the rules for PRs

Floessie, 2019-03-20:
Hi,

In #82 @wesleywesley describes the rules to obey to successfully get a PR into this project. Unfortunately they are very different from the ones outlined in [`CONTRIBUTING.md`](https://github.com/Microsemi/switchtec-user/blob/c115571634f102199b67061e5d4c2e7d952edf25/CONTRIBUTING.md). If "contributions are most welcome" `CONTRIBUTING.md` should be updated accordingly to make them happen.

Thanks,
Flössie

wesleywesley, 2019-03-20:
@Floessie 
The md file updated time was Nov 17 2017.
We have plan to update, especially for the new process, in near future.

lsgunth, 2019-03-20:
Yes, thanks for your contribution and I'm sorry in that it should not have been this difficult you. I hope they will get their procedures and documentation updated and improved so it's not such a bad experience for future contributions.

Logan

wesleywesley, 2019-04-30:
@Floessie 

The action was done by : https://github.com/Microsemi/switchtec-user/pull/86

closed 2019-05-06 by wesleywesley