Microsemi/switchtec-user #177: Executable bit set on source files

lsgunth, 2020-02-07:
I just noticed somebody is setting executable bits on source files when they should not be. These are hard to catch in reviews, can somebody clean that up and try to make sure it doesn't continue to happen:

```
-rwxr-xr-x 1 gunthorp gunthorp   3343 Feb  7 12:59 ./cli/common.h
-rwxr-xr-x 1 gunthorp gunthorp  53857 Feb  7 12:59 ./cli/main.c
-rwxr-xr-x 1 gunthorp gunthorp  32269 Feb  7 12:59 ./cli/mfg.c
-rwxr-xr-x 1 gunthorp gunthorp   1993 Feb  7 12:59 ./config.h.in
-rwxr-xr-x 1 gunthorp gunthorp   1092 Feb  7 12:59 ./configure.ac
-rwxr-xr-x 1 gunthorp gunthorp   2349 Feb  7 13:02 ./inc/switchtec/errors.h
-rwxr-xr-x 1 gunthorp gunthorp   5827 Feb  7 12:59 ./inc/switchtec/mfg.h
-rwxr-xr-x 1 gunthorp gunthorp   4627 Feb  7 12:59 ./inc/switchtec/mrpc.h
-rwxr-xr-x 1 gunthorp gunthorp  28254 Feb  7 13:05 ./inc/switchtec/switchtec.h
-rwxr-xr-x 1 gunthorp gunthorp   6902 Feb  7 12:59 ./lib/crc.c
-rwxr-xr-x 1 gunthorp gunthorp   1585 Feb  7 12:59 ./lib/crc.h
-rwxr-xr-x 1 gunthorp gunthorp  39217 Feb  7 12:59 ./lib/fw.c
-rwxr-xr-x 1 gunthorp gunthorp  21767 Feb  7 13:12 ./lib/mfg.c
-rwxr-xr-x 1 gunthorp gunthorp  15209 Feb  7 12:59 ./lib/platform/linux-i2c.c
-rwxr-xr-x 1 gunthorp gunthorp  12114 Feb  7 13:08 ./lib/platform/linux-uart.c
-rwxr-xr-x 1 gunthorp gunthorp  24161 Feb  7 13:03 ./lib/switchtec.c
-rwxr-xr-x 1 gunthorp gunthorp   7089 Feb  7 13:08 ./lib/switchtec_priv.h
-rwxr-xr-x 1 gunthorp gunthorp   5753 Feb  7 12:59 ./Makefile.in
```

xyan264, 2020-02-07:
Most likely I did that. I will clean up in a PR later.

xyan264, 2020-02-10:
Fixed in #179

closed 2020-02-10 by xyan264