Microsemi/switchtec-user #181: Disable UART access for MFG commands in BL1/2

xyan264, 2020-02-14:
'mfg' submenu commands do not support access from UART. 

Current mfg API functions give error code ERR_UART_NOT_SUPPORTED when called in MAIN boot phase, which is a good way to handle this use case. However, in BL1/2, API functions fail at 'switchtec_open', where it tries to read GAS register to get device ID. The reason is that the GAS registers are not accessible in BL1/2 through UART, so the uart_gas_read() function times out.

It would be better if 'switchtec_open' can return the same ERR_UART_NOT_SUPPORTED if UART is used in BL1/2, but this poses a chicken-and-egg issue: to return this error, we need to determine the current boot phase, and to determine the boot phase, we need to read GAS registers first...

A systemic approach might be needed to fix this issue.

lsgunth, 2020-02-14:
I don't see why this is a chicken-and-egg issue. In switchtec_open(), for the uart, try a command that should work in all firmware states, then try a command that won't work in BL2. If The first works but not the second, there's a very high probability the firmware is in an unsupported state. If both commands have retries the probability of the wrong case should be very low.

xyan264, 2020-02-14:
Yes that's a good solution, except there is no command that works in both BL1/2 & MAIN boot phase for UART. 

In fact the device UART is not even enabled in firmware when it's in BL1 mode (BL1 runs firmware from on-chip ROM, so it comes with limited capabilities).