Microsemi/switchtec-user #183: github.io API documentation out of date

xyan264, 2020-02-26:
The API page on 
   https://microsemi.github.io/switchtec-user/
only shows API version 0.9-rc2. 

@lsgunth what tools did you use to generate documents? Maybe I can use the same tool to update API documentation and push to `gh-pages`.

lsgunth, 2020-02-26:
See the doc folder in the repo. The Makefile will generate the documentation. The publish script will update the orphan branch which updates the github.io pages.

lsgunth, 2020-02-26:
Pay attention to warnings in the doxygen they indicate bad comments in the source and should be fixed.

xyan264, 2020-03-10:
@lsgunth The page: https://microsemi.github.io/switchtec-user/ is no longer accessible, is it related to what I did in the morning (below)?

This morning I tried the 'publish' script but got some error messages. And later when I tried to run the script again, it complained 'gh-pages' branch is not found. Maybe due to previous 'git push --force'?

So I went to Microsemi/switchtec-user repository and created 'gh-pages' from master branch, but I still cannot access https://microsemi.github.io/switchtec-user/

Is there some way we can fix this? I'm trying to publish from branch 'rtp-release-psx'.

lsgunth, 2020-03-10:
Not sure what you did but it looks like you made a mess of things. I don't understand how you deleted the branch. But I'm not sure exactly why it's not working now.

For starters, the gh-pages branch should be created as an orphan branch so it doesn't share the history of the main repo: 

http://stackoverflow.com/questions/13969050/ddg#13969482

Or better yet, you can probably recover the history of the original branch (at least since I last touched it) from:
https://github.com/lsgunth/switchtec-user/commits/gh-pages

The other factor you might be missing is that because you deleted the gh-pages branch, github might have disabled the website. So check that it's enabled in the projects settings (per the attached screenshot).
[image]

xyan264, 2020-03-10:
@lsgunth Thanks for the instructions. 

I have restored the 'gh-pages' (orphan) branch with history in the Microsemi/switchtec-user repo.

The https://microsemi.github.io/switchtec-user/ is not accessible yet. I don't see the 'setting' icon, in the Microsemi/switchtec-user repo, possibly it requires admin privilege. Could you check the settings for me?

Apologies for the inconvenience.

lsgunth, 2020-03-10:
I don't have admin privileges either. As I remember, someone at Microsemi held those pretty tightly. You'll have to find out who that is.

xyan264, 2020-03-10:
Thanks, I'll check with the team and find out.