Microsemi/switchtec-user #230: Failed to specify device by a PCI b:d.f

widiwei, 2020-07-16:
Hi, 
I'm trying to use the PCI bdf address instead of /dev/switchtec0, but got below the error saying "operation not supported".

Here are the logs:

_root@debian10-> switchtec version
switchtec cli version 2.0
root@debian10-> switchtec list
switchtec0                                      3.50 BF40       0000:17:00.1
switchtec1                                      3.50 BF40       0000:19:00.1
switchtec10                                     3.50 BF40       0000:ca:00.1
switchtec11                                     3.50 BF40       0000:cc:00.1
switchtec12                                     3.50 BF40       0000:e3:00.1
switchtec13                                     3.50 BF40       0000:e5:00.1
switchtec2                                      3.50 BF40       0000:31:00.1
switchtec3                                      3.50 BF40       0000:4b:00.1
switchtec4                                      3.50 BF40       0000:4e:00.1
switchtec5                                      3.50 BF40       0000:65:00.1
switchtec6                                      3.50 BF40       0000:67:00.1
switchtec7                                      3.50 BF40       0000:98:00.1
switchtec8                                      3.50 BF40       0000:b1:00.1
switchtec9                                      3.50 BF40       0000:b3:00.1
root@debian10-> switchtec info /dev/switchtec0
/dev/switchtec0:
    Generation:  GEN4
    Variant:     PFX
    Device ID:   0x4000
    FW Version:  3.50 BF40
root@debian10-> switchtec info 17:00.1
/dev/switchtec0
17:00.1: Operation not supported
root@debian10-> switchtec info 0000:17:00.1
/dev/switchtec0
0000:17:00.1: Operation not supported_

Traced the code, and I know it eventually finds the switchtecX name and use /dev/switchtecX for the operation, but somehow it didn't work. 

Is there anything I should check?

lsgunth, 2020-07-16:
I'm guessing this has something to do with having  a gen4 switch. Perhaps the generation/variant is not getting set correctly. On a quick read of the code I noticed that set_gen_variant() is probably going to be called twice due to the nested switchtec_open() and maybe that is breaking somehow. 

The code works for me, but I don't have gen4 hardware to test with. Hopefully, someone at Microsemi can reproduce it and fix it for you.

xyan264, 2020-08-31:
@widiwei 

Thank you for filing this issue. This has now been fixed and included in the new release (tag v2.4-rc3). 

You can refer to PR #232 for details about the commit that fixed this issue.

Feel free to let us know if you have other questions or concerns.

closed 2020-08-31 by xyan264