Microsemi/switchtec-user #250: Switchtec user gas read fails on 5.8.15-301.fc33.x86_64 kernel (Fedora 33)

gfxlonghorn, 2021-02-03:
I am seeing the following error when using a 5.8.15-301.fc33.x86_64 kernel and the latest switchtec user/driver:

# ./switchtec gas read 0 -a 0x2104 -b4
gas_map: Invalid argument

When we use an older kernel, we do not see this problem. In order to get the kernel driver to compile on 5.8, I had to change "pci_cleanup_aer_uncorrectable_error_status(pdev)" to "pci_aer_clear_nonfatal_status(pdev)".

lsgunth, 2021-02-03:
Hmm, this will likely take some more debugging. It's still works fine for me on 5.10. The kernel can fail a mapping with "Invalid Argument" for a number of reasons, including some security settings (like LSM modules).

First it would be good to know which mmap call is failing in linux_gas_map().

It would also useful to know if you can mmap /sys/bus/pci/devices/<switchtec_bdf>/resource0[_wc] with and without the switchtec driver loaded.

This problem is not likely related to the aer changes.

cfade, 2022-10-24:
I know this has been posted awhile ago but i believe issue might be due to iomem security restriction.

Similar to this:
https://github.com/billfarrow/pcimem/issues/13

i added iomem=relaxed to my cmdline and read/writes started to function properly.

NikS448, 2022-10-24:
Thanks for the reply! I no longer work in this field, but appreciate you taking the time.

closed 2022-10-24 by NikS448