Microsemi/switchtec-user #259: gas write/read over serial/mmapped interface not working as expected with PAX 42100

maquefel, 2021-04-08:
Hello, i am experiencing strange behavior trying to read temperature over serial/mmapped interface:

Results in zero read with no visible error:
```
./switchtec gas write /dev/switchtec0 -a 0x0 -v 0x2 -y
./switchtec gas write /dev/switchtec0 -a 0x800 -v 0x04 -y
./switchtec gas read /dev/switchtec0 -a 0x400
000400 - 0x00000000
```
```
./switchtec gas write /dev/ttyUSB0 -a 0x0 -v 0x2 -y
./switchtec gas write /dev/ttyUSB0 -a 0x800 -v 0x04 -y
./switchtec gas read /dev/ttyUSB0 -a 0x400
000400 - 0x00000000
```
Same sequence over terminal by hand gives expected result:
```
0x00000000:004f>gaswr 0x0 0x2
gas_reg_write() success
0x00000000:0050>gaswr 0x800 0x04
gas_reg_write() success
0x00000000:0051>gasrd 0x400
gas_reg_read <0x400> [1]
0x000011bd
0x00000000:0052>
```

tested on v2.6-rc1

lsgunth, 2021-04-08:
You really shouldn't be trying to do direct gas reads and writes.  Your stepping on the kernel drivers toes. Why not just use `switchtec temp /dev/switchtec0`?

maquefel, 2021-04-09:
Hello @lsgunth, temperature was taken just as an easy reproducible example.

switchtec temp uses a different mechanism than gas read/write and i am aware of it.

So are you just saying that GAS operations should be avoided and not recommended for usage ?

lsgunth, 2021-04-09:
Yes. There are a number of ways to do custom MRPC operations safely and properly in the 'examples' folder. Using the GAS interface is not a good idea.

maquefel, 2021-04-15:
Hello @lsgunth this issue is indeed is solved by #260. Is not that i really want to use gas read/write operations, but ChipLink program that i use for PAX configuration uses them in case of PCIe configuration and UART is really slow.

closed 2021-04-15 by maquefel