Microsemi/switchtec-user #279: i2c fw-update not supported

MattSimmering, 2021-08-12:
Using fw-update on i2c device fails with error code 0x11 with error message "firmware update: Programming not allowed over this interface".  It seems to fail in 'switchtec_fw_wait' in the do-while loop, after a couple iterations the 'switchtec_fw_dlstatus' returns a status of 0x11.  It seems to always fail after 4/5 iterations in this loop.

other commands like 'info', 'status', 'test', 'fw-info', etc. work on this device.  

Here's the full error: 

```
# switchtec fw-update /dev/i2c-10@0x10 ./switchtec_image.data
Writing the following firmware image to /dev/i2c-10@0x10.
File:           switchtec_image.data
Gen:            GEN4
Type:           MAP
Version:        1.0
Img Len:        0x100
CRC:            0xabcdabcd
Secure version: 0x00000000
Do you want to continue? [y/N] y
   0% [>                                                                                                                                            ] ETA:  0:26:50    5kB/s
firmware update: Programming not allowed over this interface
#
```

lsgunth, 2021-08-12:
I think the message is pretty clear: the firmware is telling us that we can't update the firmware over I2C....

MattSimmering, 2021-08-12:
So is fw-update not supported over I2C/TWI?

lsgunth, 2021-08-13:
That would be my guess. I'll let the microchip guys confirm.

xyan264, 2021-08-13:
@MattSimmering Have you tried updating individual images, such as _switchtec-bl2.fwimg_, or _switchtec-psx.fwimg_?

Also note that _switchtec-user_ does not support updating partition map image _switchtec-partmap.fwimg_ in MAIN FW boot phase.

MattSimmering, 2021-08-13:
@xyan264 Just tried this, and looked like it worked!

Is there a way to look inside the PCIe Device Information to see if the programming took?  I'd like to see things like P2P Vendor ID, P2P Device ID, etc.

xyan264, 2021-08-13:
You can use command such as `info`, `status` and `list` to find device type, FW version, and other related information. 

Also `fw-info` lists FW images on the device, so you can compare the CRC and see if it matches that of the image you updated (displayed when you update the image).

As to PCIe Device Information, _switchtec-user_ does not have a command to display these info, but there are other Linux commands for this purpose (lspci -n).

MattSimmering, 2021-08-16:
I was able to confirm programming.  Closing this issue.

closed 2021-08-16 by MattSimmering