Microsemi/switchtec-user #294: Please make release tags

morrone, 2021-11-09:
Please start making release tags in the git repo for releases. Putting a release number in just the header of a github "release" is not really sufficient. For instance, if we look at the release titled "Switchtec-user v2.6 Release for Switchtec Gen4 PFX/PSX/PAX MR3", we would see that the tag for that release is actually "v2.6-rc1", which would lead us to think it is a "release candidate" rather than a proper release". Further, the URL for that release page is "https://github.com/Microsemi/switchtec-user/releases/tag/v2.6-rc1", and the tarball in the "assets" for that release have the version as "v2.6-rc1". This makes it more difficult than it needs to be for those of us trying to build packages for these releases.

What we really want is a "v2.6" release tag, without the "rc". The github release should then be based on that. If you make the "v2.6" and annotated tag, you can put the release notes right in the commit message for the tag as well.

kelvin-cao, 2021-11-23:
Hi @morrone, 

Thanks for the suggestion. 

It makes sense to me. We will use tag names without -rcX for release in the future, just as you suggested. However, for the release note, as we will need to review it and then probably update it after it is created, so putting it in the tag commit message seems not applicable.

morrone, 2021-11-23:
That is great. Thanks!

closed 2021-11-25 by kelvin-cao