Microsemi/switchtec-user #323: info command returning "bad message"

arshad512, 2022-10-26:
I have been trying to debug an issue I am facing with "info" command with little luck. Can anyone help me chase down an issue I am facing. The details are given below and any futher info required please let me know.

Thanks

Issue.
I am getting "Bad message" on my device when issuing an "info" command
```
# ./switchtec info /dev/switchtec0
dev info: Bad message
```
Details follows

This is the version I am using.
```
# ./switchtec version
switchtec cli version 4.0
```

Device test output
```
# ./switchtec test /dev/switchtec0
test: success
```

Debugging session analysis :

My understanding is that read_resp() will have first 4 bytes as return value. Followed by variable length data.
I could see that it is returning 0x00000000, however it is still failing with -1. Where is it going wrong. 

```
(gdb) x /12x buf
0x7fffffffde70:	0x00000000	etc...
```

More detailed debugging session(important parts)

```
(gdb) s
read_resp (resp=resp@entry=0x7fffffffdf58, resp_len=resp_len@entry=8, ldev=<optimized out>) at lib/platform/linux.c:331
331	    size_t bufsize = sizeof(uint32_t) + resp_len;
(gdb) n
332		char buf[bufsize];
(gdb) 
334		ret = read(ldev->fd, buf, bufsize);
(gdb) p ldev->fd
value has been optimized out
(gdb) p bufsize
$19 = 12
(gdb) n
336		if (ret < 0)
(gdb) p ret
$20 = -1
(gdb) x /12x buf
0x7fffffffde70:	0x00000000	0x63592f39	0x00179ec8	0x00000000
0x7fffffffde80:	0xffffdf28	0x00007fff	0xf7fe76c0	0x00007fff
0x7fffffffde90:	0xffffdec0	0x00007fff	0x00000008	0x00000000
(gdb) n
353		return ret;
(gdb) n
switchtec_cmd (dev=dev@entry=0x648330, cmd=4294705408, cmd@entry=256, payload=payload@entry=0x7fffffffdf54, payload_len=payload_len@entry=4, 
    resp=resp@entry=0x7fffffffdf58, resp_len=resp_len@entry=8) at lib/platform/platform.c:174
174		if (ret > 0) {
(gdb) p ret
$26 = -1
(gdb) n
179		return ret;
(gdb) n
switchtec_get_device_info (dev=dev@entry=0x648330, phase=phase@entry=0x0, gen=gen@entry=0x0, rev=rev@entry=0x7fffffffdf9c) at lib/switchtec.c:1750
1750		if (ret == 0) {
(gdb) p ret
$27 = -1
(gdb) n
1761		} else if (ERRNO_MRPC(errno) == ERR_CMD_INVALID) {
(gdb) n
print_dev_info (dev=0x648330) at cli/main.c:203
203		if (ret) {
(gdb) p ret
$28 = -1
(gdb) n
info (argc=<optimized out>, argv=<optimized out>) at cli/main.c:232
232		return print_dev_info(cfg.dev);
(gdb) n
dev info: Bad message
205			return ret;
(gdb) 
```

lsgunth, 2022-10-26:
Hmm, very strange. Tho only place I can see that returns EBADMSG is in the kernel where it checks the status returned by the device to be valid:

https://github.com/Microsemi/switchtec-kernel/blob/e459e54607c31eabb0bb503040d23425e0d3ef9f/switchtec.c#L631

Do you know what kind of hardware and firmware version you have?

My guess is the hardware/firmware are responding improperly to the MRPC_I2C_TWI_PING command (possibly it is too old?) and thus we don't fall back to the gen3 return and it returns an error instead.

arshad512, 2022-10-26:
Thanks for your response!

>Hmm, very strange. Tho only place I can see that returns EBADMSG is in the kernel where it checks the status returned by the device to be valid:

I will check this part. Thanks. However, I am curious to understand. Is not the return value 0 here ? I may be wrong.

```
(gdb) x /12x buf
0x7fffffffde70:	0x00000000	0x63592f39	0x00179ec8	0x00000000
0x7fffffffde80:	0xffffdf28	0x00007fff	0xf7fe76c0	0x00007fff
0x7fffffffde90:	0xffffdec0	0x00007fff	0x00000008	0x00000000
```

> Do you know what kind of hardware and firmware version you have?

```
# cat /sys/class/switchtec/switchtec0/fw_version 
01090063
# cat /sys/class/switchtec/switchtec0/vendor_id 
MICROSEM
```

>My guess is the hardware/firmware are responding improperly to the MRPC_I2C_TWI_PING command (possibly it is too old?) >and thus we don't fall back to the gen3 return and it returns an error instead.

This is what I saw a while back when debugging. I thought I should share it here. It looks to recognize "GEN3" correctly. I am not sure if this is what you are mentioning here.

```
(gdb) p ldev
$5 = (struct switchtec_linux *) 0x648330
(gdb) p *ldev
$6 = {dev = {device_id = 34098, gen = SWITCHTEC_GEN3, var = SWITCHTEC_PFX, pax_id = -1, local_pax_id = -1, partition = 0, partition_count = 1, 
    boot_phase = SWITCHTEC_BOOT_PHASE_FW, name = "/dev/switchtec0", '\000' <repeats 4080 times>, gas_map = 0x0, gas_map_size = 0, ops = 0x644080 <linux_ops>}, fd = 3}
```

lsgunth, 2022-10-26:
The return value in the buffer is zero. But it's not likely valid if the status is the wrong value.

Is your hardware indeed GEN3? The switchtec_dev stuff you are printing may not be correct given the switchtec_get_device_info() call failed.

What is in:

/sys/class/switchtec/switchtec0/component_id

arshad512, 2022-10-27:
>What is in:
>/sys/class/switchtec/switchtec0/component_id

```
# cat /sys/class/switchtec/switchtec0/component_id
PM8532
```
>Is your hardware indeed GEN3? The switchtec_dev stuff you are printing may not be correct given the switchtec_get_device_info() call failed.

I thought the switchtec_get_device_info() is already populated before calling switchtec_cmd()/linux_cmd(). I will revisit my understanding again.

lsgunth, 2022-10-27:
Yup, this is definitely GEN3 hardware. Perhaps a patch such as this will fix the issue:

```diff
diff --git a/lib/switchtec.c b/lib/switchtec.c
index 23b570fc8d30..4640532f5f00 100644
--- a/lib/switchtec.c
+++ b/lib/switchtec.c
@@ -1755,7 +1755,7 @@ int switchtec_get_device_info(struct switchtec_dev *dev,
                        *rev = (dev_info >> 8) & 0x0f;
                if (gen)
                        *gen = map_to_gen((dev_info >> 12) & 0x0f);
-       } else if (ERRNO_MRPC(errno) == ERR_CMD_INVALID) {
+       } else if (errno == EBADMSG || ERRNO_MRPC(errno) == ERR_CMD_INVALID) {
                if (phase)
                        *phase = SWITCHTEC_BOOT_PHASE_FW;
                if (gen)
```

arshad512, 2022-10-27:
Your patch worked. Thanks @lsgunth !

```
# ./switchtec info /dev/switchtec0
/dev/switchtec0:
    Generation:  GEN3
    HW Revision: Unknown
    Variant:     PFX
    Device ID:   0x8532
    FW Version:  1.09 B063

```

This issue can be closed.

PR #324 (2022-10-27): lib: Fallback on EBADMSG error code
Some GEN3 hardware ends up responding to MRPC_I2C_TWI_PING with a bad status code and the kernel translates into an EBADMSG error code. When this happens, the code should fall back into the GEN3 path.

Resolves: #323 ("info command returning 'Bad message'")
commit cbe7f8cc (2022-10-27): lib: Fallback on EBADMSG error code

Some GEN3 hardware ends up responding to MRPC_I2C_TWI_PING with a bad
status code and the kernel translates into an EBADMSG error code.
When this happens, the code should fall back into the GEN3 path.

Resolves: #323 ("info command returning 'Bad message'")
closed 2022-10-27 by kelvin-cao