Microsemi/switchtec-user #325: events command returning "Invalid argument"

arshad512, 2022-10-28:
Issue:
Running "events" command reports "Invalid argument"

Expected:
Should work successfully and not throw ("Invalid argument")

Please let me know if any further information/debuging info is required. 

Thanks

Output currently seen:
```
# ./switchtec events /dev/switchtec0
event_ctl: Invalid argument
```

**Details follows**

Switchtec version

```
# ./switchtec version
switchtec cli version 4.0
```
Hardware and firmware details

```
# cat /sys/class/switchtec/switchtec0/fw_version 
01090063
# cat /sys/class/switchtec/switchtec0/vendor_id 
MICROSEM
# cat /sys/class/switchtec/switchtec0/component_id
PM8532
```
Debugging session(important parts)
linux_event_ctl()  passes for SWITCHTEC_PART_EVT_PART_RESET and SWITCHTEC_PFF_EVT_TLP_THROTTLING. It fails for SWITCHTEC_PFF_EVT_AER_IN_P2P

```
Breakpoint 1, linux_event_ctl (dev=0x648390, e=SWITCHTEC_PART_EVT_PART_RESET, index=0, flags=0, data=0x0) at lib/platform/linux.c:906
906		if (e >= SWITCHTEC_MAX_EVENTS)

Breakpoint 1, linux_event_ctl (dev=0x648390, e=SWITCHTEC_PFF_EVT_TLP_THROTTLING, index=0, flags=0, data=0x0) at lib/platform/linux.c:906
906		if (e >= SWITCHTEC_MAX_EVENTS)
(gdb) c
Continuing.

Breakpoint 1, linux_event_ctl (dev=0x648390, e=SWITCHTEC_PFF_EVT_AER_IN_P2P, index=197, flags=0, data=0x0) at lib/platform/linux.c:906
906		if (e >= SWITCHTEC_MAX_EVENTS)
(gdb) c
Continuing.
event_ctl: Invalid argument
```

lsgunth, 2022-10-28:
After staring at the code for a bit, I think the problem is with the legacy event summary ioctl (SWITCHTEC_IOCTL_EVENT_SUMMARY_LEGACY) fallback path.

I'm assuming you are using an older kernel module that is falling back to this. If you are using a kernel older than about v5.2 then you will be exercising the fallback code. Please confirm.

When the fall back code is used, the switchtec_event_summary structure is not fully populated seeing the legacy structure returned by the kernel is smaller. This combined with the fact that it is not initialized in the CLI means you might have spurious bits set in the structure.

The spurious bit is set and it causes an incorrect linux_event_ctl call with an index of 197 (which I believe is too high for your hardware) and the kernel ioctl rightly returns EINVAL when trying to query an event with that high of an index.

The solution is hopefully simple: to just ensure the summary structure is fully zeroed.  There are a number of call sites of switchtec_event_summary() so I think it's best to ensure it is zeroed in event_summary_copy(). You can try this untested patch:

If this doesn't work, please do some debugging around what's being returned in linux_event_summary().

```diff
diff --git a/lib/platform/linux.c b/lib/platform/linux.c
index 25ba8fd14cf8..76c03d645ba6 100644
--- a/lib/platform/linux.c
+++ b/lib/platform/linux.c
@@ -829,6 +829,9 @@ static void event_summary_copy(struct switchtec_event_summary *dst,
 
        for (i = 0; i < SWITCHTEC_MAX_PFF_CSR && i < size; i++)
                dst->pff[i] = src->pff[i];
+
+       for (; i < SWITCHTEC_MAX_PFF_CSR; i++)
+               dst->pff[i] = 0;
 }
 
 #define EV(t, n)[SWITCHTEC_ ## t ## _EVT_ ## n] = \
```

arshad512, 2022-10-29:
Thanks @lsgunth !
I will try out your patch and let you know.

arshad512, 2022-10-29:
My quick test passes. Your patch works. Thanks a log @lsgunth . (However, please do not close this issue right now) I will let you know after I futher verify.

**Quick test**: Passes

```
# ./switchtec events /dev/switchtec0
Partition 0 Events:
	PART_RESET            	1   	Partition Reset
    Port 0:
	TLP_THROTTLING        	255 	TLP Throttling Event
```

arshad512, 2022-10-31:
@lsgunth , For one of the devices I was getting ... (Power Management Event). At this point I do not know if this slipped in or is actually an event.
  ```
  Port 0:
	POWER_MGMT            	1   	Power Management Event
	TLP_THROTTLING        	255 	TLP Throttling Event
```
I was double checking the bits under switchtec_ioctl_event_summary (pff (partation)) and found to be correctly getting the values. 9th and 10th bit enabled (ie two bits enabled). Versus just 1 bit enabled ( when POWER_MGMT is not showing).

Please feel free to close this issue. Thanks again!

commit 9349ef63 (2022-11-08): lib/platform/linux: Ensure entire switchtec_event_summary struct is set

In normal cases, event_summary_copy() will set the entire
switchtec_event_summary struct. However, if the library falls back to
the smaller SWITCHTEC_IOCTL_EVENT_SUMMARY_LEGACY, then a size less than
SWITCHTEC_MAX_PFF_CSR will be passed to the function and a portion
of the function will be left unset.

There are a number of places in the code where this function gets
called pointing to an unitialized structure on the stack. So fix
this issue by ensuring event_summary_copy() sets the entire
structure.

Resolves: #325 ("eve [... cut]
closed 2022-11-15 by kelvin-cao