Microsemi/switchtec-user #333: TWI Recovery in WSL2

Sarikano, 2023-12-11:
Hey guys,

do you think its possible to access TWI recovery over Windows somehow?

I tried to access via WSL2 and switchtec kernel. I needed to bind the USB-I2C device to the WSL2 but cannot get a connection to the device.
[image]
Do you think there is a possible workaround ?

lsgunth, 2023-12-13:
I don't have any experience with WSL2 but it sounds like the kernel doesn't have I2C support, or the I2c module is not loaded. You could try 'modprobe i2c-dev', failing that you may have to compile a kernel with CONFIG_I2C_CHARDEV set and possibly also a driver for the I2C device you are using. 

I wouldn't be surprised if the WSL2 kernel has no I2C built in, but I have no idea.

Sarikano, 2023-12-13:
Hey,

Thanks for your comment.
Meanwhile i found out that the wsl2 kernel does indeed not hav the i2c module. So this is something i can confirm.

Currently trying to build my own kernel with it.

Will keep you on tracking.

Cheers